OLIMEX/ESP8266 #15: Compile, then error ;-(

ly0, 2015-06-30:
```
latyas@zypher:~/ESP8266-master/IoT Firmware$ make
make[1]: Entering directory `/home/latyas/ESP8266-master/IoT Firmware/olimex'
make[2]: Entering directory `/home/latyas/ESP8266-master/IoT Firmware/olimex/user'
make[3]: Entering directory `/home/latyas/ESP8266-master/IoT Firmware/olimex/user/modules'
xt-xcc -Os -g -O2 -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals  -DICACHE_FLASH   -I include -I ./ -I ../ -I ../../include/ets -I ../include -I ./ -I ../../include/ets -I ../../include -I ../../../include -I ../../../include/eagle  -o .output/eagle/debug/obj/mod_finger.o -c mod_finger.c
mod_finger.c: In function 'void finger_buff_init(uint32, uint8, uint16)':
mod_finger.c:17:65: error: 'pvPortZalloc' was not declared in this scope
   finger_buff = (finger_packet *)os_zalloc(sizeof(finger_packet));
                                                                 ^
mod_finger.c:25:28: error: 'vPortFree' was not declared in this scope
   os_free(finger_buff->data);
                            ^
mod_finger.c:31:25: error: 'pvPortZalloc' was not declared in this scope
   (uint8 *)os_zalloc(len)
                         ^
In file included from mod_finger.c:2:0:
mod_finger.c: In function 'void finger_send_buff()':
../../../include/osapi.h:39:41: error: 'os_printf_plus' was not declared in this scope
  os_printf_plus(flash_str, ##__VA_ARGS__); \
                                         ^
../stdout.h:4:16: note: in expansion of macro 'os_printf'
  #define debug os_printf
                ^
mod_finger.c:48:2: note: in expansion of macro 'debug'
  debug("\nFINGER: Sending packet...\n");
  ^
../../../include/osapi.h:39:41: error: 'os_printf_plus' was not declared in this scope
  os_printf_plus(flash_str, ##__VA_ARGS__); \
                                         ^
../stdout.h:4:16: note: in expansion of macro 'os_printf'
  #define debug os_printf
                ^
mod_finger.c:49:2: note: in expansion of macro 'debug'
  debug("Address: 0x%08x\n", finger_buff->address);
  ^
../../../include/osapi.h:39:41: error: 'os_printf_plus' was not declared in this scope
  os_printf_plus(flash_str, ##__VA_ARGS__); \
                                         ^
../stdout.h:4:16: note: in expansion of macro 'os_printf'
  #define debug os_printf
                ^
mod_finger.c:50:2: note: in expansion of macro 'debug'
  debug("PID: 0x%02x\n", finger_buff->pid);
  ^
../../../include/osapi.h:39:41: error: 'os_printf_plus' was not declared in this scope
  os_printf_plus(flash_str, ##__VA_ARGS__); \
                                         ^
../stdout.h:4:16: note: in expansion of macro 'os_printf'
  #define debug os_printf
                ^
mod_finger.c:51:2: note: in expansion of macro 'debug'
  debug("Len: 0x%04x\n", finger_buff->len);
  ^
../../../include/osapi.h:39:41: error: 'os_printf_plus' was not declared in this scope
  os_printf_plus(flash_str, ##__VA_ARGS__); \
                                         ^
../stdout.h:4:16: note: in expansion of macro 'os_printf'
  #define debug os_printf
                ^
mod_finger.c:52:2: note: in expansion of macro 'debug'
  debug("Data: ");
  ^
../../../include/osapi.h:39:41: error: 'os_printf_plus' was not declared in this scope
  os_printf_plus(flash_str, ##__VA_ARGS__); \
                                         ^
../stdout.h:4:16: note: in expansion of macro 'os_printf'
  #define debug os_printf
                ^
mod_finger.c:55:3: note: in expansion of macro 'debug'
[... 122 lines cut ...]
mod_finger.c:210:3: note: in expansion of macro 'debug'
   debug("FINGER: Device found\n");
   ^
../../../include/osapi.h:39:41: error: 'os_printf_plus' was not declared in this scope
  os_printf_plus(flash_str, ##__VA_ARGS__); \
                                         ^
../stdout.h:4:16: note: in expansion of macro 'os_printf'
  #define debug os_printf
                ^
mod_finger.c:212:3: note: in expansion of macro 'debug'
   debug("FINGER: Device not found\n");
   ^
mod_finger.c: In function 'void finger_init()':
mod_finger.c:232:36: error: 'setTimeout' was not declared in this scope
  setTimeout(finger_check, NULL, 500);
                                    ^
make[3]: *** [.output/eagle/debug/obj/mod_finger.o] Error 1
make[3]: Leaving directory `/home/latyas/ESP8266-master/IoT Firmware/olimex/user/modules'
make[2]: *** [.subdirs] Error 2
make[2]: Leaving directory `/home/latyas/ESP8266-master/IoT Firmware/olimex/user'
make[1]: *** [.subdirs] Error 2
make[1]: Leaving directory `/home/latyas/ESP8266-master/IoT Firmware/olimex'
make: *** [.subdirs] Error 2

```

peter-valkov, 2015-06-30:
You have to use following command sequence to compile firmware

    make clean
    make COMPILE=gcc SPI_SIZE=2048 BOOT=new APP=1

...this will generate bin/upgrade/user1.1024.new.bin 
 
    make clean
    make COMPILE=gcc SPI_SIZE=2048 BOOT=new APP=2

...this will generate bin/upgrade/user2.1024.new.bin 

Then you should power your ESP8266 while holding button pressed to enter in bootloader mode.
Command for uploading firmware is:

    esptool/esptool.py --baud 576000 write_flash 0x00000 bin/boot_v1.2.bin 0x01000 bin/upgrade/user1.1024.new.bin 0x81000 bin/upgrade/user2.1024.new.bin --flash_size 16m

peter-valkov, 2015-06-30:
Let me know if this resolves your problem

ly0, 2015-06-30:
Problem solved. thanks :-)

closed 2015-06-30 by peter-valkov