OLIMEX/ESP8266 #21: Togle mode for relay

bobybc, 2015-09-24:
Hi,
does it possible to have 3-th state for relay "toggle" , which should switch it "on" and after a while switch back to "off". 
The goal is to have "button" like behavior with only one HTTP request . 
Now it is possible with two separated requests. 

10x
Boby

peter-valkov, 2015-09-24:
"toggle" for me have different meaning from what you described. It is (current_state XOR 1) i.e. if it is OFF switch it ON - if it is  ON switch it OFF. 

Back to your question: 
1. there is need for additional parameter - delay after it will be switched OFF
2. how the following situation should be handled:
- we have request for "ON-OFF" 
- before delay for switching OFF is expired we receive another "ON-OFF" request
- should we switch it OFF after the delay of first request is expired or should we wait for the second?!?

There can be arguments for either solution.

Why two requests bother you? 

On-OFF is easy implemented using JavaScript setTimeout() function.

```javascript
function relaySet(socket, state) {
	socket.send(
		JSON.stringify(
			{
				URL: '/relay',
				Method: 'POST',
				Data: {
					Relay: state
				}
			}
		)
	);
}

relayOnOff(socket, delay) {
	relaySet(socket, 1);
	setTimeout(
		function () {
			relaySet(socket, 0);
		},
		delay
	);
}

var socket = new WebSocket('ws://192.168.4.1/events');
relayOnOff(socket, 500);

```

bobybc, 2015-09-25:
I don't use browser, but Android application to open my automated door. It happens that in a slow mobile connection (GPRS) the delay b/w ON and OFF requests is more than 3 sec, even 5 sometimes. This reflects on a different mode for my door  - it switches to so called pedestrian mode where the door is not fully open. 
That's why i need to have ON-OFF with one request... 

And let we simplify the case - currently you code has setRelay(1)/setRelay(0)  to ON/OFF relay.
Can we just put delay interval (in ms) as parameter to this function? 
Server side it can be implemented like : if the value is greater than 1, just switch ON, wait  time in ms described in the value and then switch OFF.
What do you think on that?

peter-valkov, 2015-09-25:
I prefer to make relay state signed integer so if the state:
- is equal to 0 - switch OFF 
- is greater than 0 - switch ON 
- is lower than 0 - switch ON wait abs(state) ms then switch OFF

What about the case with double requests? 
- receive request for ON-OFF after 500ms
- switch relay ON
- 200ms later second request is received for new 500ms

What should I do? 
- switch relay OFF 500ms after first request
- switch relay OFF 500ms after second request i.e. 700ms after first request

bobybc, 2015-09-25:
The short answer is - switch relay OFF 500ms after first request
But I think that it will be good if it is synchronized i.e. until first request is not completed, do not execute new one

peter-valkov, 2015-09-25:
@bobybc your proposal is done. Now you can set the relay state as follows:

- If [state] is 0 relay is switched OFF.
- If [state] is 1 relay is switched ON.
- If [state] is greater than 1 relay is switched ON for [state]ms and then switched OFF. During this interval all other commands are ignored.

Hope your door will work as expected ;-)

closed 2015-09-25 by peter-valkov

bobybc, 2015-09-25:
10x a lot Peter
Cheers :)