OLIMEX/ESP8266 #24: ESP8266-EVB EMI noise

yerpj, 2015-11-21:
I encountered some odd behaviour with the EVB. I mounted an http server on the ESP8266 ( javascript thanks to Espruino), and I tried to switch the relay state everytime a client connects to the server (thus, giving me a audible debug info). The case happened, where the ESP reseted by itself after the relay state changed. I experienced this many times.
I think it may come from the electro-magnetic field created by the relay inductor, which may be important enough to cause electrical errors on the ESP side.
 Does anyone experienced the same behaviour?

toome123, 2015-11-23:
I think the problem is not hardware . but you will tell you what your consumer touch with relay?

closed 2015-12-09 by yerpj

toome123, 2015-12-13:
One day ago , I had the same problem, I upload simple sketch to switch relay over mqtt . And after five toggles of relay state, the esp module, block and restarting self

toome123, 2015-12-13:
Is there really a chance that this is a hardware problem or memmory leak

DanKoloff, 2015-12-14:
Hey guys,

We have tested the relays numerous times in different scenarios. They also get tested during and after the assembly of each board. I just tested to switch the relays a hundrded times and the behavior described by you does not occur here. What I recommend you to do:

1) Test with another (better) power supply adapter.
2) Always add software delays when switching relays. Switching relays on and off without any delay would greatly shorten their lifespan and, in fact, can damage the relay in a matter of minutes.
3) Consider that the issue might be related to the load on the relay. Refer to the information below.

The whole thing might be the typical problem with all electromechanical relays. Inductive AC and DC loads are hard for electomechanical relays where welding contacts cause problems. Normal way to solve this problem is to use clamp diodes or zener diodes in the poles of the load. However, this lengthens the off delay and it does not totally eliminate the sparking in relay contacts. Switching the high DC voltages is also difficult for the electromechanical relays.

I am sure there is a lot of solutions that might be suggested online. There are a number of workarounds or partial solutions but they depend on your exact design and setup. There is no solution that I can recommend 100% - you will have to try different things.

I have few ideas if the problem is related to the load (point 2):

- Try to place the ESP8266-EVB board further from any motors. If you use a 220V motor the whole freeze might be related to its influence on the ESP8266-EVB.

- Try to place a TVS varistor/Zener near the power supply of ESP8266-EVB - the varistor picked should be suitable for [<your input voltage> + 3], e.g. if you supply ESP8266 with 5V the TVS should be for 8V.

- If your motor commutates DC (not AC) you can also put capacitors in parallel to the contacts.

- If nothing else works check if there are suitable solid-state relays that can be used instead of the one we have placed.

Best regards,
Lub/OLIMEX

yerpj, 2015-12-14:
Thank you @DanKoloff for those recommendations. 
I encountered this issue without any load on the relay. My power supply is a 5V 3A, and I always put a 0.5sec delay before switching the relay. Anyway, I never try to make PWM with a mechanical relay ;-) It is switched at most a few times per minute.

I do not know if this issue comes from hardware or software , but in any case it seems that both @toome123 and me are experiencing some odd behaviors with that board.
I am using [Espruino](https://github.com/espruino) on it, and I know there is a few bugs with the port on ESP8266. @toome123 , which kind of firmware are you using?

DanKoloff, 2015-12-14:
Can you test with our IoT firmware (in this repository)? This would show if it is software-related issue. There is a category called "Relay" in the web form. Search for "relay" here: https://github.com/OLIMEX/ESP8266/tree/master/IoT%20Firmware

Also check this video (despite that it doesn't show the usage of the on-board relay; you can see the "Relay" category in the web form): https://www.youtube.com/watch?v=4zRq85791vo

toome123, 2016-05-23:
The problem I had I think was the supply is not enough. @yerpj I use Olimex IoT firmware.  I think this issue can be closed.