OLIMEX/ESP8266 #25: Build error

donvercety, 2015-11-26:
```sh
/bin/sh: 2: xt-xcc: not found
xt-xcc -Os -g -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections  -DICACHE_FLASH   -I include -I ./ -I ../ -I ../../include/ets -I ../include -I ./ -I ../../include/ets -I ../../include -I ../../../include -I ../../../include/eagle  -o .output/eagle/debug/obj/mod_emtr.o -c mod_emtr.c
make[3]: xt-xcc: Command not found
```

mikexmas, 2015-11-26:
Maybe wrong path
I run "make COMPILE=gcc SPI_SIZE_MAP=3 BOOT=new APP=1" 
which uses gcc and works for me (running on Windows)

donvercety, 2015-11-26:
Thanks, `SPI_SIZE_MAP=3 BOOT=new` this helped me.
Also the documentation lacks some information...
There is a `DEVICE` parameter in the make instruction example, but `DEVICE` is not defined anywhere in the make file.

my working solution: `make DEVICE=SWITCH2 SSL=0 COMPILE=gcc SPI_SIZE_MAP=3 BOOT=new APP=1 EXTRA_CCFLAGS='-DDEVICE=SWITCH2 -DSSL=0 -DCOMPILE=gcc -DSPI_SIZE_MAP=3 -DBOOT=new -DAPP=1'`

`.2048.new.3.bin successully in folder bin/upgrade.` :)

closed 2015-11-26 by donvercety

peter-valkov, 2015-11-26:
Sorry. Forgot to sync make file.  Now should be OK.