OLIMEX/ESP8266 #28: reboot while esp8266 connect to the AP.

Paulxia, 2016-01-10:
WiFi Scan: 8 APs found.
f r0, WiFi Scan: Event Raise.
scandone
f 0, WiFi Scan: Auto Detect...
scandone
state: 0 -> 2 (b0)
station: 08:11:96:f7:04:d8 leave, AID = 1
rm 1
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2

connected with MWIFI-9RS8, channel 11
dhcp client start...
cnt 
ip:192.168.99.169,mask:255.255.255.0,gw:192.168.99.1
EVENTS: Station connected
Fatal exception 29(StoreProhibitedCause):
epc1=0x4020c03d, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000

Here is the map

4020c02c <webclient_new_request>:
	char *host, 
	int port, 
	char *path, 
	char *headers, 
	char *content
) {
4020c02c:	c0c112        	addi	a1, a1, -64
4020c02f:	f109      	s32i.n	a0, a1, 60
4020c031:	e1c9      	s32i.n	a12, a1, 56
4020c033:	d1d9      	s32i.n	a13, a1, 52
4020c035:	c1e9      	s32i.n	a14, a1, 48
4020c037:	b1f9      	s32i.n	a15, a1, 44
4020c039:	3179      	s32i.n	a7, a1, 12
4020c03b:	017d      	mov.n	a7, a1
4020c03d:	1f29      	s32i.n	a2, a15, 4
	}
	debug("WEBCLIENT: Requests count: %d\n", count);
#endif
#endif
	
	STAILQ_FOREACH(request, &webclient_requests, entries) {
4020c03f:	fe4f21        	l32r	a2, 4020b97c <crc16+0x34>
	char *host, 
	int port, 
	char *path, 
	char *headers, 
	char *content

peter-valkov, 2016-01-21:
> WiFi Scan: 8 APs found

Can you please test it in less crowded AP environment, as I am unable to reproduce the problem?

Paulxia, 2016-01-22:
Less crowded AP environment is OK.  
I found a temp solution.
LOCAL char events[1024];

void ICACHE_FLASH_ATTR user_event_raise(char *url, char *data) {
	if (data == NULL || os_strlen(data) == 0) {
		return;
	}
	
	uint16 event_size = 
		WEBSERVER_MAX_VALUE + 
		(url == NULL ? 0 : os_strlen(url)) + 
		os_strlen(data)
	;
	#if 0
	char events[event_size];
	user_event_build(events, url, data);
	#else
	user_event_build(events, url, data);
	#endif

peter-valkov, 2016-01-27:
FIX: Event messages memory allocation https://github.com/OLIMEX/ESP8266/commit/5aeca72462667f0b01fc0a2581244cee24260ed9

closed 2016-01-27 by peter-valkov