OLIMEX/ESP8266 #32: IoT  Firmware Websocket "freezes" on frame segmentation at TCP level

sdrsdr, 2016-01-21:
Attached [oliiot.tcpdump.zip](https://github.com/OLIMEX/ESP8266/files/99108/oliiot.tcpdump.zip) is a tcpdump from web socket session to one of your devices. After the HTTP handshake a TCP packet is send with 6 bytes - the header of a websocket frame, then another TCP packet is sent with the playload. Your client fails to acknowledge this packets and then some re transmission  happens and practically no communication happens.  As TCP is stream oriented  protocol it is perfectly posible to receive every byte in a message in separate TCP packet and this should not break your "server"

sdrsdr, 2016-01-21:
Issue found with a project using http://wslay.sourceforge.net/ for websocket framing

peter-valkov, 2016-01-21:
Thank you for collaboration! I just committed a fix without be able to reproduce exact situation. Can you please test it in your environment and tell me if it works.

Regards,
Peter

sdrsdr, 2016-01-21:
My current firmware update tab says "current: user2.bin " and  "boot: v2 normal" I'm not sure if I can update both userX.bin and whic boot.bin should I use .. there is no v2 in  your bin directory

peter-valkov, 2016-01-21:
If on *Firmware Update* tab **Current** says **user2.bin** you should upload firmware for **user1.bin** which is ***user1.2048.new.3.bin*** located in bin/upgrade folder 

and vice versa 

If **Current** says **user1.bin** you should upload **user2.bin** which is ***user2.2048.new.3.bin*** located in same folder

sdrsdr, 2016-01-21:
I'm unable update the firmware ..  after just few 1.4k  packets exchanged the network traffic fails to resends  and no nothing happens :(

closed 2016-01-21 by sdrsdr

sdrsdr, 2016-01-21:
Also non websocket update seems to succeed but the web-page does not reconnect back fully - some of the request get responded some  get 404  but  no other tabs except "ESP Connection" appears :(

peter-valkov, 2016-01-22:
If over the air firmware update fails you can update it with serial cable.