OLIMEX/ESP8266 #33: Enhancement: Upgrade SDK to 1.5.1 (16.1.2016) (IoT Firmware)

mikexmas, 2016-01-22:
I noticed some troubles working with softAP; to connect my comp to softAP: connected and 
immediately disconnected.
station: 00:27:10:81:40:f4 join, AID = 1
station: 00:27:10:81:40:f4 leave, AID = 1
rm 1
add 1
aid 1
station: 00:27:10:81:40:f4 join, AID = 1
station: 00:27:10:81:40:f4 leave, AID = 1
rm 1
add 1
aid 1
station: 00:27:10:81:40:f4 join, AID = 1
station: 00:27:10:81:40:f4 leave, AID = 1
rm 1
etc.....

I upgraded to new SDK and it may be much better.
Just copied SDK files to working folder, except "makefile".

peter-valkov, 2016-01-22:
In release notes on espressif site:

> 4. Optimize softAP working mode so that some stations can be easily connected.

... so probably it will help. First thing in my next week TODO list will be to test it.

peter-valkov, 2016-01-25:
SDK is updated to version 1.5.1. In the quick test I used boot-loader version 1.5. In my environment it seems everything works fine so far. If you continue to experience trouble connecting to SoftAP we might need to address the issue to espressif with more detailed information about the hardware used as station to connect to SoftAP.

closed 2016-01-25 by peter-valkov