OLIMEX/ESP8266 #34: IoT Firmware: Default config settings after complete flashing

mikexmas, 2016-01-22:
When ESP is complete flashed (also blank.bin and esp_init_data_default.bin) there is different configuration of softAP & station which is stored by ESP SDK than when you press long button and restore_defaults occurs.

Is there a way to detect initial ESP settings in user_main to be overriden by restore_defaults (which works better)  not only with long-button press?

Best Regards

peter-valkov, 2016-01-22:
Press the button and hold it for more than 5 secs. It will reset settings to IoT Firmware defaults.

peter-valkov, 2016-01-22:
Here is the command I use for complete flash 
> esptool/esptool.py --baud 576000 write_flash 0x00000 bin/boot_v1.2.bin 0x01000 bin/upgrade/user1.2048.new.3.bin 0x81000 bin/upgrade/user2.2048.new.3.bin 0x1FC000 bin/esp_init_data_default.bin 0x1FE000 bin/blank.bin --flash_size 16m

peter-valkov, 2016-01-22:
Note the last blank.bin at address 0x1FE000

peter-valkov, 2016-01-22:
If you want to clear previous IoT Firmware settings as well you should flash with blank.bin at address 0x100000

peter-valkov, 2016-01-22:
... so complete flash command should look like this:

> esptool/esptool.py --baud 576000 write_flash 0x00000 bin/boot_v1.2.bin 0x01000 bin/upgrade/user1.2048.new.3.bin 0x81000 bin/upgrade/user2.2048.new.3.bin 0x100000 bin/blank.bin 0x1FC000 bin/esp_init_data_default.bin 0x1FE000 bin/blank.bin --flash_size 16m

peter-valkov, 2016-01-25:
After update to SDK 1.5.1 and new bootloader version 1.5 complete firmware flash command should look like:

> esptool/esptool.py --baud 576000 write_flash 0x00000 bin/boot_v1.5.bin 0x01000 bin/upgrade/user1.2048.new.3.bin 0x81000 bin/upgrade/user2.2048.new.3.bin 0x100000 bin/blank.bin 0x1FC000 bin/esp_init_data_default.bin 0x1FE000 bin/blank.bin --flash_size 16m

@mikexmas if this solves your problem let me know so we can close this issue.

Best Regards
Peter

closed 2016-01-27 by peter-valkov