OLIMEX/ESP8266 #42: Tho functions move into flash instead of RAM

mikexmas, 2016-05-04:
Is there any special reason to have following two fncs in RAM ?
user_config.c/config_response
user_webclient.c/webclient_host_check

I stray of RAM iram1_0_seg since SDK 1.5.3.

peter-valkov, 2016-05-05:
There is no reason above mentioned functions to be in RAM. Just an oversight on my part. Fix is ready and I will commit it later today. Thank you for the cooperation.

However I have no problems with compilation. What options you use to compile?

peter-valkov, 2016-05-05:
Resolved with commit 606458377ba7534763f4a992455a39a2357a3a72

mikexmas, 2016-05-05:
I use Olimex MAKEFILE, some additional C/H files added only with my code. Size of .text increases to cca 8250hex, but limit is 8000hex.
I checked DUMP, but no my code is placed into .text, so maybe some additional system calls are done, but I have not investigated yet.
I added to linker options: -Wl,--gc-sections to decrease size.
This happened after migration to 1.5.3.

peter-valkov, 2016-05-05:
I will recommend you to compile with make DEVICE=EVB_ONLY APP=1 (or 2 depending which binary you want to compile). This will reduce module drivers to minimum and will leave more RAM for your code. 

You can look at user_config.h to see drivers included for each preset device configurations.

At the moment device presets are: EVB_ONLY, PLUG, SWITCH1, SWITCH2, BADGE, DIMMER.

closed 2016-05-06 by peter-valkov