OLIMEX/ESP8266 #43: after commit SDK 2.0.0 IoT Firmware won't to compile

NMCity, 2016-07-27:
xtensa-lx106-elf-gcc -Os -g -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections  -DICACHE_FLASH -DDEVICE=0 -DSSL=0 -DBUILD=\"\"   -I include -I ./ -I ../include -I ../../include -I ../../include/eagle  -o .output/eagle/debug/obj/i2c_master.o -c i2c_master.c
xtensa-lx106-elf-ar ru .output/eagle/debug/lib/libdriver.a .output/eagle/debug/obj/hw_timer.o .output/eagle/debug/obj/uart.o .output/eagle/debug/obj/gpio16.o .output/eagle/debug/obj/pwm.o .output/eagle/debug/obj/i2c_extention.o .output/eagle/debug/obj/spi.o .output/eagle/debug/obj/key.o .output/eagle/debug/obj/i2c_master.o 
xtensa-lx106-elf-ar: creating .output/eagle/debug/lib/libdriver.a
make[2]: Leaving directory '/home/chris/Dokumente/ESP8266/IoT Firmware/olimex/driver'
xtensa-lx106-elf-gcc  -L../lib -nostdlib -T../ld/eagle.app.v6.new.1024.app1.ld -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group -lc -lgcc -lhal -lphy -lpp -lnet80211 -llwip -lwpa -lmain -ljson -lupgrade -lssl -lsmartconfig -lcrypto user/.output/eagle/debug/lib/libuser.a driver/.output/eagle/debug/lib/libdriver.a user/modules/.output/eagle/debug/lib/libmod.a -Wl,--end-group -o .output/eagle/debug/image/eagle.app.v6.out 
/opt/Espressif/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: .output/eagle/debug/image/eagle.app.v6.out section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
../Makefile:371: recipe for target '.output/eagle/debug/image/eagle.app.v6.out' failed
make[1]: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1
make[1]: Leaving directory '/home/chris/Dokumente/ESP8266/IoT Firmware/olimex'
Makefile:316: recipe for target '.subdirs' failed
make: *** [.subdirs] Error 2

maybe this belongs to changes in eagle.app.v6  ???

peter-valkov, 2016-07-28:
> `.text' will not fit in region `iram1_0_seg'

Looks like because of new added SDK functionality your code can not fit into region size.
Try with following commands for user1.bin and user2.bin

```
make DEVICE=EVB_ONLY APP=1
make DEVICE=EVB_ONLY APP=2
```
this will reduce count of drivers for our modules to the minimum so hopefully your code will fit into region.

Let me know if this help.

NMCity, 2016-07-28:
i have just make a fresh clone of the repo with the following result:

```bash
xtensa-lx106-elf-ar ru .output/eagle/debug/lib/libdriver.a .output/eagle/debug/obj/hw_timer.o .output/eagle/debug/obj/uart.o .output/eagle/debug/obj/gpio16.o .output/eagle/debug/obj/pwm.o .output/eagle/debug/obj/i2c_extention.o .output/eagle/debug/obj/spi.o .output/eagle/debug/obj/key.o .output/eagle/debug/obj/i2c_master.o 
xtensa-lx106-elf-ar: creating .output/eagle/debug/lib/libdriver.a
make[2]: Leaving directory '/home/chris/Dokumente/ESP8266/IoT Firmware/olimex/driver'
xtensa-lx106-elf-gcc  -L../lib -nostdlib -T../ld/eagle.app.v6.new.1024.app1.ld -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group -lc -lgcc -lhal -lphy -lpp -lnet80211 -llwip -lwpa -lmain -ljson -lupgrade -lssl -lsmartconfig -lcrypto user/.output/eagle/debug/lib/libuser.a driver/.output/eagle/debug/lib/libdriver.a user/modules/.output/eagle/debug/lib/libmod.a -Wl,--end-group -o .output/eagle/debug/image/eagle.app.v6.out 
/opt/Espressif/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: .output/eagle/debug/image/eagle.app.v6.out section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
../Makefile:371: recipe for target '.output/eagle/debug/image/eagle.app.v6.out' failed
make[1]: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1
make[1]: Leaving directory '/home/chris/Dokumente/ESP8266/IoT Firmware/olimex'
Makefile:316: recipe for target '.subdirs' failed
make: *** [.subdirs] Error 2
```

peter-valkov, 2016-07-28:
1. Do you try to compile firmware with or without modifications?
2. What is the exact command sequence you use to compile the source?

NMCity, 2016-07-28:
**without modificatons**

```bash

mkdir test
cd test
git clone https://github.com/OLIMEX/ESP8266.git
cd ESP8266/IoT\ Firmware
make clean
make DEVICE=1 APP=1
```

**Result**

```bash
make[2]: Leaving directory '/home/chris/test/ESP8266/IoT Firmware/olimex/driver'
xtensa-lx106-elf-gcc  -L../lib -nostdlib -T../ld/eagle.app.v6.new.1024.app1.ld -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group -lc -lgcc -lhal -lphy -lpp -lnet80211 -llwip -lwpa -lmain -ljson -lupgrade -lssl -lsmartconfig -lcrypto user/.output/eagle/debug/lib/libuser.a driver/.output/eagle/debug/lib/libdriver.a user/modules/.output/eagle/debug/lib/libmod.a -Wl,--end-group -o .output/eagle/debug/image/eagle.app.v6.out 
/opt/Espressif/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: .output/eagle/debug/image/eagle.app.v6.out section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
../Makefile:371: recipe for target '.output/eagle/debug/image/eagle.app.v6.out' failed
make[1]: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1
make[1]: Leaving directory '/home/chris/test/ESP8266/IoT Firmware/olimex'
Makefile:316: recipe for target '.subdirs' failed
make: *** [.subdirs] Error 2
```

peter-valkov, 2016-07-29:
Strange i have no such problems. The only difference I see is gcc  and crosstool version

```
Using built-in specs.
COLLECT_GCC=xtensa-lx106-elf-gcc
COLLECT_LTO_WRAPPER=/home/tefan/workspace/esp-open-sdk/xtensa-lx106-elf/libexec/gcc/xtensa-lx106-elf/4.8.2/lto-wrapper
Target: xtensa-lx106-elf
Configured with: /home/tefan/workspace/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/configure --build=x86_64-build_unknown-linux-gnu --host=x86_64-build_unknown-linux-gnu --target=xtensa-lx106-elf --prefix=/home/tefan/workspace/esp-open-sdk/xtensa-lx106-elf --with-local-prefix=/home/tefan/workspace/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot --disable-libmudflap --with-sysroot=/home/tefan/workspace/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG 1.20.0' --disable-__cxa_atexit --with-gmp=/home/tefan/workspace/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-mpfr=/home/tefan/workspace/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-mpc=/home/tefan/workspace/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-isl=/home/tefan/workspace/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-cloog=/home/tefan/workspace/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-libelf=/home/tefan/workspace/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --enable-lto --enable-target-optspace --disable-libgomp --disable-libmudflap --disable-nls --disable-multilib --enable-languages=c,c++
Thread model: single
gcc version 4.8.2 (crosstool-NG 1.20.0) 
```

NMCity, 2016-07-29:
I try to find a solution with the toolchain ( maybe a eraliery commit as a workaround).
I think i was read something about a similar problem in unofficial DevKit for windows

peter-valkov, 2016-09-23:
Today I ran into same problem when installing toolchain to different location. It turned out that for some reason two of the library files are missing. Hope that the issue is fixed with commit 48b948b9f38d6d614bd70ebd69548f91edc0e803 

Can you check if this works for you?

closed 2016-10-11 by peter-valkov

davydnorris, 2016-11-01:
I had same problem, and discovered that there is a version of libgcc.a supplied in the SDK with numerous functions removed that are in the esp8266 ROM. If your built libgcc.a is in your path first, it will include these functions and the result will be too large.

The attached **_xtensa-lx106-elf-ar_** script will remove the functions from the built library and then your programs will compile.

Use it as follows:
`export PATH=$(TOOLCHAIN)/bin:${PATH}
cd $(TOOLCHAIN)/lib/gcc/xtensa-lx106-elf/<version>
xtensa-lx106-elf-ar -M < strip_libgcc_funcs.txt`

[strip_libgcc_funcs.txt](https://github.com/OLIMEX/ESP8266/files/563190/strip_libgcc_funcs.txt)